nu-radio/NuRadioMC #600: generator.py does not take energy/flavor dependence into account when generating ccnc, inelasticities

sjoerd-bouma, 2023-12-05:
In `generator.py`, when 'randomly' deciding whether a CC or NC interaction takes place, the code uses the `get_ccnc` function [here](https://nu-radio.github.io/NuRadioMC/_modules/NuRadioMC/utilities/inelasticities.html#get_ccnc). This corresponds to a constant ratio of CC : NC events, even though the ratio actually depends on the energy and the particle flavor. Weirdly, parameterizations for the cross section that do take this into account (as well as the slightly more significant dependence on neutrino vs antineutrino) are already available in the [NuRadioMC.utilities.cross_sections](https://nu-radio.github.io/NuRadioMC/NuRadioMC/apidoc/NuRadioMC.utilities.cross_sections.html#NuRadioMC.utilities.cross_sections.get_nu_cross_section) module. While the effect is relatively small, it seems strange not to take it into account, especially if the appropriate code is already there? 

On a similar note, the neutrino inelasticity is also generated independently of the neutrino energy - I'm guessing for similar reasons.
[image]

sjoerd-bouma, 2023-12-06:
Potentially relevant in this context (?) - #396

cg-laser, 2023-12-07:
yes, this is still a todo for me to finally finish #396 ... the missing piece is to use the new BGR18 cross sections everywhere consistently

sjoerd-bouma, 2024-03-18:
Just as a gentle reminder, driven by the discussion started by https://arxiv.org/abs/2403.07984 - I think it's important to remember that until #396 gets merged, our inelasticities are off by somewhere between 10%, a factor of 2, or an order of magnitude, depending on the energy and how we define the comparison. This is not currently clearly documented (in my opinion), but should be kept in mind when running effective volume simulations.
[image]

cg-laser, 2024-03-21:
We already did extensive tests about the impact of using BGR18 dsigma/dy distributions vs. the current default for the cross section paper. (https://arxiv.org/abs/2204.04237) In that paper, we simulated the Veff as a function of shower energy for nu_e CC interactions and other interactions that only produce a hadronic shower. These Veff are then folded with the dsigma/dy distributions and the a flux to get event rates. We compared this to our standard procedure of simulating the Veff for fixed neutrino energies with the older cross sections (and fixed CC/NC fractions). The effect on event rates was less than 10%. 
I should #396 but I already know that the impact will not be large.

sjoerd-bouma, 2024-03-21:
Thanks, that's good to know - from the magnitude of the difference as shown in the above plots I definitely would have guessed that the difference, particularly at higher neutrino energies, would have been larger than that. Conversely, it probably means the impact of https://arxiv.org/abs/2403.07984 will be negligible in terms of effective volume.

fschlueter, 2025-08-08:
This has been completed with #396