nu-radio/NuRadioMC #616: Interface to modify RNO-G detector description

fschlueter, 2024-02-05:
This was mentioned in one of the RNO-G calls. Several analyses (I guess mostly calibration related) will profit / make it necessary to change the detector description "on-the-fly" to validate the effect on the changed description. I can think of a few approaches how to realise that. This issue is here to collect ideas and as a discussion platform. 

I have two ideas:
- Implement a setter for the `total_response` of a channel. Keep in mind this key is buffered internally. Also this key is not exported to file (because one can/should not serialise custom classes and the data are already exported). 
- Implement a setter for the signal chain list which specifies the individual response which constitute the `total_reponse` (those would than need to be defined in the database)

fschlueter, 2024-02-05:
A other possibility is to modify the json file. If that is the case we should add some utility scripts to NuRadio for that.

cg-laser, 2024-02-09:
what about a new class "ModifiedDetector" that inherits from the RNO-G detector but allows the user to overwrite any parameters/response functions? This detector class could also implement other useful functions such as systematically changing response functions to study systematic uncertianties, etc. 
It might be good to separate this functionality from the RNO-G detector class to let users unintentionally mess up the reference detector description.

cg-laser, 2024-02-09:
and as always, I would take a pragmatic approach and implement only the functionality that someone needs for their analysis right now. If more functionality is needed at a later stage, this can be added later.

closed 2025-05-02 by fschlueter