nu-radio/NuRadioMC #623: .run() function not working with mattak update (36b3b90dcba270cd98e41b0f3bfbd4eca756af9d)

sanyukta13, 2024-02-09:
The current version of rno-g mattak provides an error related to wfs upon running .run() on read root file. Fix I used was to git pull the version before this update. The changes in mattak Dataset.py and consequent dataset files resulted in this issue.

fschlueter, 2024-02-09:
Hi @sanyukta13 thanks for spotting it and already providing a short term solution. I will have a look when I am back from my vacation. Sorry for the inconvenience

sjoerd-bouma, 2024-02-09:
This is a mattak issue, I think. As far as I can tell, the ROOT backend works fine, but the uproot backend gives the error
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[12], line 1
----> 1 for evt in reader.run():
      2     for station in evt.get_stations():
      3         print(evt.get_id(), station.get_station_time())

File ~/Python/software/nu-radio/NuRadioMC/NuRadioReco/modules/io/RNO_G/readRNOGDataMattak.py:786, in readRNOGData.run(self)
    784 # Just read wfs if necessary
    785 if wfs is None:
--> 786     wfs = dataset.wfs()
    788 waveforms_of_event = wfs[idx]
    790 evt = self._get_event(evtinfo, waveforms_of_event)

File ~/Python/software/RNO-G/mattak/py/mattak/backends/uproot/dataset.py:246, in Dataset.wfs(self, calibrated)
    244 w = None 
    245 if self.full or self.skip_incomplete: 
--> 246     w = self._wfs['radiant_data[24][2048]'].array(entry_start=self.first, entry_stop=self.last, library='np')
    247 elif not self.multiple: 
    248     if self.first in self.events_with_waveforms: 

AttributeError: 'Dataset' object has no attribute '_wfs'"
```

sanyukta13, 2024-02-09:
Yes it was missing 'Dataset' in '_wfs', I'll add it on the mattak repo. Thanks!

sjoerd-bouma, 2024-02-13:
I think Cosmin fixed this with https://github.com/RNO-G/mattak/commit/265d0435c812c06a46b9f32c2f99eb904f8307f5, so I'm closing this issue. Thanks for flagging it @sanyukta13 !

closed 2024-02-13 by sjoerd-bouma

fschlueter, 2024-02-13:
The issue was resolved with a fix in the mattak repo [commit](https://github.com/RNO-G/mattak/commit/265d0435c812c06a46b9f32c2f99eb904f8307f5). @sanyukta13 thanks a lot for taking care of this!

fschlueter, 2024-02-13:
Lol Sjoerd was faster than me. Apparently I have to refresh my browser more often.