nu-radio/NuRadioMC #639: io `coreas.make_sim_station()`: factorize out CoREAS to NRR efield conversion to improve flexibility

twybo, 2024-02-26:
**Describe the issue**
In `NuRadioReco.modules.io.coreas.coreas`, the `make_sim_station` method takes an hdf5 and hdf5 observer to make a `sim_station`. In this process, it takes the responsibility to convert from the CoREAS to NRR format ( `Ex, Ey = -Ey, Ex` and cgs to SI).

**Expected behavior**
The above is responsibility at the cost of flexibility. This could be shifted to the user who would input onsky NRR electric fields. The conversion of CoREAS to NRR format could be factorized out as a convenience function also in `coreas.py`. Should this be handled with a flag in `make_sim_station` to not lose its convenient converting behaviour, or moved outside altogether?

**Additional context**
This came up when building a CoREAS reader that interpolates a detector layout within a starshape (`readCoREASInterpolator` in the lofar/interpolator branch). Sequence of events: NRR conversion &rarr; interpolation &rarr; `sim_station` creation.

cg-laser, 2024-02-27:
what is an "NRR" electric field? 
The module is made to read in raw coreas files, therefore, the automatic conversion needs to remain. However, it is totally fine to make this conversion optional (but the default should be to do it), and to refactor to have the conversion in a central utility function in coreas.py if it is needed in several places.

twybo, 2024-02-29:
With nrr electric field I mean an electric field array in nuradioreco's geomagnetic coordinate system (correct axes ordering, as expected for `radiotools.coordinatesystems.cstrafo` conversions) with field strengths in si units.
I'll refactor and keep the default behaviour with an option to opt out.

closed 2024-02-29 by twybo