nu-radio/NuRadioMC #657: Reading normal Detector from nur file not working

MijnheerD, 2024-03-13:
**Describe the issue**
When trying to load a normal Detector (i.e. not a Generic one) from a `nur` file, an error is thrown. The cause is line 389 in NuRadioRecoio.py, where the `__new__` method is called. At some point (I believe this happened in commit 844f3aa9be081da98b62bd593a40da9cfaa52114 , where the concept of GenericDetector was introduced) the `json_filename` parameter in this method changed from an optional to required parameter. So now the call on line 389 is missing a required parameter.

Interestingly, the GenericDetector does not produce this error because its `__new__` method is not overwritten.

**Additional context**
MWE to reproduce the error:
```python
import NuRadioReco.modules.io.eventReader.

reader = NuRadioReco.modules.io.eventReader.eventReader()
reader.begin('path/to/nur', read_detector=True)
detector = reader.get_detector()
```

MijnheerD, 2024-03-13:
I see two solutions here:

1. Add an empty string as the `json_filename` on line 389 in NuRadioRecoio.py -> in the next line, when `__init__` is called, we use a dictionary as source anyway
2. Rethink the `__new__` method of the `Detector` class. Nor `GenericDetector` nor `DetectorBase` re-implement this method. Is it really necessary?

fschlueter, 2024-03-13:
I think we can just use the function `NuRadioReco.detector.Detector` to create the detector object

fschlueter, 2024-03-13:
@sjoerd-bouma, I would rather use the `detector.Detector` function!

sjoerd-bouma, 2024-03-13:
> @sjoerd-bouma, I would rather use the `detector.Detector` function!

Feel free to hijack my branch (or make your own), I agree that would be the even cleaner solution : ) Was just curious if this would solve the issue

fschlueter, 2024-03-13:
@MijnheerD can you share a file for testing?

MijnheerD, 2024-03-13:
@fschlueter I'll send you on Slack, I don't think Github would be happy with a 220MB file 🙃

MijnheerD, 2024-03-13:
@sjoerd-bouma To satisfy your curiosity I tried it, and it indeed removes the error ;-)

PR #658 (2024-03-25): Hotfix/detector from nur
Fixes #657.
closed 2024-03-25 by fschlueter