nu-radio/NuRadioMC #671: Too many CoREAS readers

anelles, 2024-04-10:
This is just to keep track of the issue and collect the discussion of how to move forward. 

@lpyras, @MijnheerD, @twybo have been working on different CoREAS readers recently and in particular with the interpolation function in https://github.com/nu-radio/cr-pulse-interpolator there seem to be a issue in reconciling where the modularity is kept. 

Currently every reader also contains part of the analysis approach (i.e. reading just one station, reading random stations, interpolating to a given detector) and the question is whether we could modularize further to avoid duplication of the general reading functionality that parses the CoREAS files and generating the interpolator object (if so desired).

anelles, 2024-04-10:
@cg-laser this seems like a discussion that once needs to be had during a developers call.

lpyras, 2024-05-31:
This will be addressed in PR #685