nu-radio/NuRadioMC #702: Definition of Fresnel coefficients

philippwindischhofer, 2024-07-31:
Hi all,

_(First of all, I have only recently started using NuRadioMC and my question below might simply be a consequence of me misunderstanding the code or its coordinate system. Apologies if that is the case.)_

My question is: does anybody happen to have a reference for where the Fresnel coefficients `r_p` and `r_s` (defined [here](https://github.com/nu-radio/NuRadioMC/blob/a56e0b0cc9106a0bad32069f8e5c60bb543491fa/NuRadioReco/utilities/geometryUtilities.py#L144-L171)) have been taken from? I am wondering particularly about the overall sign of `r_p` (where e.g. Jackson differs from NuRadioMC), and its phase shift in the case of total internal reflection. I attach more information below.

1. Sign of `r_p`: Looking at Eq. 7.60 in Jackson (page 220 [here](http://www.fisica.ugto.mx/~ggutj/CV/Classical_Electrodynamics_Jackson_1a_Edicion.pdf)) and converting to the same notation used in NuRadioMC, there seems to be an overall sign difference relative to NuRadioMC. The same is true for the definition given on [Wikipedia](https://en.wikipedia.org/wiki/Fresnel_equations#Complex_amplitude_reflection_and_transmission_coefficients). If anybody remembers the original reference that was used in the code, I'd love to take a look at it to understand this difference.

2. Phase shift for total internal reflection: Again taking Jackson's derivation for concreteness, he uses the "negative frequency" `$\exp(-i*\omega*t)$` convention (e.g. Eq. 7.49 on page 217 [here](http://www.fisica.ugto.mx/~ggutj/CV/Classical_Electrodynamics_Jackson_1a_Edicion.pdf)). In NuRadioMC, the Fresnel coefficients are applied as [multiplicative factors](https://github.com/nu-radio/NuRadioMC/blob/a56e0b0cc9106a0bad32069f8e5c60bb543491fa/NuRadioMC/SignalProp/analyticraytracing.py#L2752-L2761) to the Fourier spectra of the signals. The spectra are [calculated](https://github.com/nu-radio/NuRadioMC/blob/a56e0b0cc9106a0bad32069f8e5c60bb543491fa/NuRadioReco/utilities/fft.py#L33-L44) through `np.rfft`, which returns the "positive-frequency" components (see documentation [here](https://numpy.org/doc/stable/reference/generated/numpy.fft.rfft.html) and sign convention [here](https://numpy.org/doc/stable/reference/routines.fft.html#implementation-details)). To apply the correct phase shift, shouldn't one then use the complex conjugate of the Fresnel coefficients? This only matters in cases with total internal reflection, of course. Again, whether or not complex conjugation should be applied is easy to establish if we can look up the sign conventions used in the original reference.

Please let me know what you think! Tagging @sjoerd-bouma and @cg-laser, who touched this code before. I'm probably missing other people who have worked on this before.

Cheers,
Philipp

philippwindischhofer, 2024-08-04:
Here is what the above would look like: https://github.com/nu-radio/NuRadioMC/compare/develop...philippwindischhofer:NuRadioMC:phwindis_fresnel

philippwindischhofer, 2024-08-05:
Hi Christian!

Thanks for digging out that old discussion! I had searched around, but couldn't find anything---I had totally overlooked that archived repository. I'll read up on the discussion there.

> It seems that your solution (inverting the sign of r_p + adding the complex conjugate) will leave the phase shift for TIR invariant for the p-polarization? Then it will be compatible with the ARIANNA DnR data.

Correct, for cases (like TIR) where `r_p` is mostly purely imaginary, the sign will not change as a result of the changes I'm proposing here.

Cheers,
Philipp

cozzyd, 2024-08-05:
Note that the sign of the Fresnel Coefficients is (oddly) a convention. See e.g: https://ieeexplore.ieee.org/document/9184440

Though it's entirely possible you already know this and I'm missing the point of confusion.

philippwindischhofer, 2024-08-06:
Hi Cosmin, if I understand correctly, NuRadioMC uses what your reference calls the "antiparallel `E_i_p` & `E_r_p`" convention. My point of confusion is that the corresponding Fresnel coefficient appears to be sign-flipped relative to what I would expect it to be in this convention (but maybe I misunderstand the code?)

philippwindischhofer, 2024-08-08:
Hi again, just to add a quick note about the situation that Dave suggested to look at in the call earlier today.

At normal incidence, the "parallel" and "perpendicular" polarizations (w.r.t. the plane of incidence) both become parallel to the surface, and for `n_2 > n_1` the reflected wave should pick up a sign flip (as the reflection happens off of the "optically denser" medium).

Using the sign convention on the sketch below, this means we should expect `r_p > 0` and `r_s < 0` at normal incidence.

[image]
[image]

philippwindischhofer, 2024-10-02:
@cg-laser I just wanted to check back briefly to see if you have had any chance to think about this? How do you think we should proceed?

philippwindischhofer, 2024-10-24:
I did a quick comparison with the fully-electrodynamic signal (calculated with Eisvogel) in a situation where it should agree with geometric optics.

Eisvogel being an experimental code, that comes with its own caveats and grains of salt, of course. FWIW, the result is below: after the change proposed here, the signal polarities are in agreement for both TIR and non-TIR cases, whereas before that was not the case.

[image]
[image]

cg-laser, 2024-10-24:
this looks very good! Thanks for the comparison. 
Did you use MEEP to calculate the Greensfunction for Eisvogel? Or what was the underlying signal propagation code?

philippwindischhofer, 2024-10-24:
Yes, the Green's function comes from MEEP!

sjoerd-bouma, 2025-03-06:
Closed by #708, thanks @philippwindischhofer !

closed 2025-03-06 by sjoerd-bouma