nu-radio/NuRadioMC #737: Unnecessary ffts in BaseTrace class

fschlueter, 2024-10-21:
When looking at the `base_trace.py` class it seems to me that we unnecessarily "resetting" either the time domain trace or frequency spectrum with the goal of "to much security". 

In the getter functions `get_trace()` and `get_frequency_spectrum()` we set the other representation to None. This of course avoids to store "the same data twice" but unless we perform any operation on either of them we have to perform unnecessary FFTs if we back and forth request both representations. Since we only can change the data with the setter function (we only return copies of the arrays in the getter functions) it would be sufficient to only set the other representation to None in the setter functions. 

However, as I said this will increase memory consumption. Is that a limiting factor and the reason for the current implementation or did I miss something?
Maybe @cg-laser, @shallmann  or @sjoerd-bouma

cg-laser, 2024-10-21:
we should loop in @christophwelling . We added more security due to unintended changes by reference that could occur otherwise.

fschlueter, 2024-10-21:
I see, but I think the copy calls in the getter functions should prevent this. We could also add unit tests to catch this

fschlueter, 2025-02-11:
@christophwelling any comment? So to clarify. I am not proposing to remove any copy statement for additional security. I am proposing to keep the time domain and frequency domain in memory and only set one to None if we have modified the other (which we can nicely control by the getter functions).

christophwelling, 2025-02-11:
I'm okay with that change as long as we are careful. Memory consumption should be okay for things like RNO-G, but for larger experiments like SKA I think you might run into some problems.
One option could be to have 2 getter functions, one that returns a copy and keeps both representations, and one that returns a reference and will reset the other representation. This might be overkill though.

closed 2025-08-08 by fschlueter