nu-radio/NuRadioMC #792: Store `antenna_by_depth` in detector json

sjoerd-bouma, 2025-01-17:
The `Detector` class on initialization has the annoying default argument `antenna_by_depth=True` even though this results in unusable antenna names for most detectors I have seen in the past 4 years. While we could change this and probably save a lot of inexperienced users some headaches at the cost of technically breaking backwards-compatibility, whether or not antennas are stored as full antenna descriptions or as short strings where the antenna depth suffix should be automatically added is a property of the json, not something that is dynamic at class initialization. I therefore think we should move towards just storing this value inside the detector jsons.

This should be a relatively straightforward change.

fschlueter, 2025-01-17:
I absolutely agree. I think it caused me headaches twice (because I forgot it after the first time). Would be great to avoid it in the future!

cg-laser, 2025-01-28:
I agree, we can change the default parameter.