nu-radio/NuRadioMC #823: Air-to-ice / ice-to-air raytracing is broken

sjoerd-bouma, 2025-02-04:
After a tentative implementation of air-to-ice or ice-to-air raytracing using the (analytic) raytracer in #558, this feature was silently removed in #660.

At least in the function `get_delta_y` some additional code will have to be (re)implemented to be able to use this feature again. This will require taking some more of the internal helper functions of `ray_tracing_2d` and making them standalone functions, so they can be compiled using numba.

alisanozdrina, 2025-11-19:
I attempted to bring back air-to-ice or ice-to-air raytracing based on #558 and the comment above: https://github.com/nu-radio/NuRadioMC/compare/develop...raytracing_air-to-ice.  So far it reproduces results from #558. Cant compare focusing calculation since some issues from #649 has been since fixed. Only work with numba, use_cpp flag has to be False. 

[image]

fschlueter, 2025-11-20:
Thats great @alisanozdrina, do you want to report on that in todays MC call at 3pm CET?

anelles, 2026-03-27:
Maybe this can also be fixed in part by #1077 -- at least I have seen plots that show that in-ice raytracing already works.