nu-radio/NuRadioMC #829: Serialising parameters

fschlueter, 2025-02-11:
Currently we are storing parameters in a dictionary with keys which are strings are compost of `class_name.parameter_name`. I do not like this to much because I) it does not allow us to change names (which is typically possible for enums because internally they are treated as integer) and II) it is in principle inefficient (we can just use the integer value) (yes yes that is not a crucial bottleneck). Can we store them as tuples, (class, enum_int)? How the class is stored I am actually not sure, probably as string?

fschlueter, 2025-02-11:
Any thoughts @sjoerd-bouma or @cg-laser

sjoerd-bouma, 2025-02-17:
Do we want to be able to rename enums? If we ever move to a 'better' storage format than .nur (e.g. hdf5) we will probably store them by name, not by int, I would have thought. Also, while in principle we can store things internally however we want, every change will have to be kept and checked for to maintain read-in compatibility, which means the reader code will grow and become more complicated. So my first response is that I'm not sure if the benefits outweigh the costs on this.

fschlueter, 2025-02-17:
I just do not like the fact that we are stuck with bad names forever. I certainly do not want to change the meaning of enums just give them better descriptive names. I do understand your argument with a different file format ...

cg-laser, 2025-02-18:
the identifier for the parameters are their names, which also means that we can't change the name. If a name turns out to be bad, we need to remove it completely and add a new quantity with a different name. 
Identifying parameters by int is very intransparent. The parameter storage is essentially a dict, but through the enums we can make sure to have well-defined, unique names. 
I don't think we should change anything about that.

closed 2025-02-18 by fschlueter

sjoerd-bouma, 2025-02-18:
One enum can have multiple names, so if there are any you feel particularly strongly about, I would be open to adding some aliases (though again this might complicate things long-term somewhat for the same reason as given above). You would have to check how this works with the dictionary-like access in ParameterStorage but this could of course be updated if necessary.