nu-radio/NuRadioMC #836: eventWriter broken with ElectricFields

MijnheerD, 2025-02-20:
**Describe the issue**
After the changes in #831 , the eventWriter fails to write out ElectricFields. The issue seems to be that in line 43 of `_fastnumpyio.py` a sum is done between the `array.data` and some other information. However, these are of two different types which are incompatible. Namely,
```python
>>> import numpy as np
>>> ar = np.random.uniform(size=(3,200))
>>> type(ar)
<class 'memoryview'>
```
The other elements on line 43 are of type `bytes`, which is incompatible. Hence the error:
```python
TypeError: can't concat memoryview to bytes
```

My instinct tells my to add a simple cast and `+ bytes(array.data)` at the end of line 43, but I am not sure if this messes with other things.

**Expected behavior**
Well, it should not crash :)

**Additional context**
This issue came up in the context of the refactoring of the CoREAS modules, in PR #688 . There is one weird thing about this issue, and that is that I cannot reproduce this error by manually creating an ElectricField and adding it to some Event. Somehow when saving this Event , it saves like normal. So it seems like sometimes the type of `array.data` is bytes already?

MijnheerD, 2025-02-20:
To add to my last comment in the issue, what I mean is this. This piece of code does not trigger the error.

```python
import numpy as np

from NuRadioReco.framework.electric_field import ElectricField
from NuRadioReco.framework.event import Event
from NuRadioReco.framework.station import Station
from NuRadioReco.modules.io.eventWriter import eventWriter

field = ElectricField([0,1])
field.set_trace(np.random.uniform(size=(3, 200)), 1)

station = Station(0)
station.add_electric_field(field)

evt = Event(0, 0)
evt.set_station(station)

writer = eventWriter()
writer.begin('test.nur')
writer.run(evt)
```

Even though

```python
>>> type(field.get_trace().data)
<class 'memoryview'>
```

MijnheerD, 2025-02-20:
> How do you get an array to be a `memoryview`?

The array is not a `memoryview`, rather the data attribute is. This appears to be the default behaviour and is a way to avoid copying data (from my understanding, a `memoryview` appears to be kind of like a read-only pointer).

MijnheerD, 2025-02-20:
So I have really delved deep and I am at a loss. I have two scripts, which produce `ElectricField` objects with the same shape. Both are Numpy arrays, all the different type checks amount to exactly the same. But still, one gives the error and the other does not. The only relevant section I could find in the Python docs is 
> bytes-like object
An object that supports the [Buffer Protocol](https://docs.python.org/3/c-api/buffer.html#bufferobjects) and can export a C-[contiguous](https://docs.python.org/3/glossary.html#term-contiguous) buffer. This includes all [bytes](https://docs.python.org/3/library/stdtypes.html#bytes), [bytearray](https://docs.python.org/3/library/stdtypes.html#bytearray), and [array.array](https://docs.python.org/3/library/array.html#array.array) objects, as well as many common [memoryview](https://docs.python.org/3/library/stdtypes.html#memoryview) objects.

It says "many common memoryview objects" . So my only guess here is that somehow the two Numpy arrays produce different types of memoryview objects. At this point I have no idea what to test further.

I can confirm that making the change I suggested in the original issue (ie casting to `bytes` explicitly) does solve the issue, so I would still suggest this as the fix.

MijnheerD, 2025-02-20:
> Maybe this would mean that somehow in the case where the error pops up, the entire array is not C-contiguous? Hence making it not a valid bytes-like object, hence not being able to be joined with the rest of the bytes array? However I have no idea how to test/fix this.

Okay, this was way easier than I thought 🙈 Just doing
```python
array.flags['C_CONTIGUOUS']
```
shows that indeed the array is not C-contiguous when the ElectricField has been constructed in `readCoREASDetector()` .

MijnheerD, 2025-02-20:
Okay, sorry for the spam. Last message, I promise!

I finally have a example script that reproduces the error:
```python
import numpy as np

from NuRadioReco.framework.electric_field import ElectricField
from NuRadioReco.framework.event import Event
from NuRadioReco.framework.station import Station
from NuRadioReco.modules.io.eventWriter import eventWriter

field = ElectricField([0,1])
field.set_trace(np.random.uniform(size=(4082, 3)).T, 1)

station = Station(0)
station.add_electric_field(field)

evt = Event(0, 0)
evt.set_station(station)

writer = eventWriter()
writer.begin('test.nur')
writer.run(evt)
```
Notice the transform when setting the field trace. This is the culprit! When the array is initialised, it is C-contiguous. However, when transposing it it becomes F-contiguous (to learn more about this, I found [this answer](https://stackoverflow.com/questions/26998223/what-is-the-difference-between-contiguous-and-non-contiguous-arrays) very insightful). The reason we _have_ to transpose though, is that the `set_trace()` method only accepts the traces with the samples on the second axis.

This now also explains why this only popped up when working with the interpolator. The problem is that the interpolator produces the traces with the samples _on the first axis_, which means that we have to transpose them when adding them to the `ElectricField` objects. This then breaks the C-contingency (is this the right word?) of the array. And hence it leads to this problem.

Now, then to the solution. We could use a function like [this](https://numpy.org/doc/1.26/reference/generated/numpy.ascontiguousarray.html) to convert the array to have the correct memory layout. But I am not sure where to inject this exactly (maybe in the `set_trace()` function as we know it needs to work well with the eventWriter?). Probably it is easier to do the casting to bytes, although I guess we should then check that the data is not altered in this process?

As a sidenote, is there a way in which we could add this as a testcase in `test-io.py` ? To prevent later modifications to the I/O code to run into the same issue.

fschlueter, 2025-02-20:
🤿

sjoerd-bouma, 2025-02-20:
Thanks @MijnheerD for finding this and identifying a solution so quickly! The simplest and most generic fix would be to call `np.ascontiguousarray` inside `_fastnumpio.pack`, which would be a one-line fix. But it's late so I will do this tomorrow.

sjoerd-bouma, 2025-03-03:
Closed via #839 / #688

PR #839 (2025-03-03): Make numpy arrays contiguous before storing them
Fixes #836 

@MijnheerD found that the _fastnumpyio implementation to store numpy arrays only works for contiguous arrays, and raises an error otherwise. So I've added `np.ascontiguousarray` to the _fastnumpyio function, which copies the array data to contiguous ordering if this is not already the case.
closed 2025-03-03 by sjoerd-bouma