nu-radio/NuRadioMC #837: Add module to store Event as CoREAS-like HDF5

MijnheerD, 2025-02-20:
When interacting with CoREAS simulations, we are converting HDF5 files to the NUR format. However, in some cases it might be useful to store them back as HDF5 files (for example when interpolating a simulation). It would be nice if you (yes, you can do it, I believe in you!) added a module that could do this conversion.

fschlueter, 2025-08-08:
Haha, love that motivation :) @MijnheerD maybe you can find someone from the air-shower crowd to do it? It's not quite so relevant in the context of RNO-G (for SKA I can clearly see the appeal of this feature!)

MijnheerD, 2025-08-11:
@galaxyvita has some experience with converting stuff to HDF5, so he will take a shot at this :-)