nu-radio/NuRadioMC #846: repository size is very big

cozzyd, 2025-02-26:
**Describe the issue**
A clone uses > 500 MB of disk space, most of which is not the current tree. This seems to be largely due to now-deleted binary files. 

```
$ git rev-list --objects --all | git cat-file --batch-check='%(objecttype) %(objectname) %(objectsize) %(rest)' | sed -n 's/^blob //p' | sort --numeric-sort --key=2 | tail -n 20 | cut -c 1-12,41- | $(command -v gnumfmt || echo numfmt) --field=2 --to=iec-i --suffix=B --padding=7 --round=nearest
6a5c039b33fd  6.4MiB NuRadioMC/test/SingleEvents/1e18_output_reference.nur
c63e979b0250  6.6MiB NuRadioMC/test/SingleEvents/1e18_output_reference.nur
0a65a6d780ac  6.6MiB NuRadioMC/test/SingleEvents/1e18_output_reference.nur
f2eeb6e7d04f  7.1MiB simulation/event_input/1e18_n1e5.hdf5
8812d65e4e24  8.1MiB NuRadioReco/modules/io/noise/Untitled.ipynb
5a9e0842e3ec  9.9MiB simulation/eventlist_test02.txt
b75f9b8a7b26  9.9MiB simulation/eventlist_test01.txt
7a3b7f7cbe9e   14MiB NuRadioMC/examples/01_Veff_simulation_Aishwarya/rnog_detector_station24.json
ed8c46e37f1c   15MiB NuRadioReco/examples/cr_efficiency_analysis/air_shower_sim/SIM000472.hdf5
99ded74ebd70   15MiB NuRadioReco/examples/cr_efficiency_analysis/air_shower_sim/SIM000479.hdf5
a03229bd7a60   15MiB NuRadioReco/examples/cr_efficiency_analysis/air_shower_sim/SIM000485.hdf5
1069de72b194   15MiB NuRadioReco/examples/cr_efficiency_analysis/air_shower_sim/SIM000495.hdf5
764a2e2b71a1   20MiB NuRadioMC/test/atmospheric_Aeff/1e18eV/1e18_full.hdf5
1f0b6c8b8c42   22MiB NuRadioMC/examples/aniso_raytracing/dummy_sim/dummy_outp_ARA02_aniso.nur
9f42101cf45f   23MiB NuRadioReco/modules/io/noise/windy/windy_512.npy
a135c9352139   27MiB NuRadioReco/examples/cr_efficiency_analysis/output_air_shower_reco/air_shower_reco_pb_80_180_tt_2.7e-05.nur
9682d2019f43   34MiB NuRadioMC/test/SignalProp/reference_birefringence.npy
41f3dea60b85   35MiB NuRadioMC/examples/05_pulser_calibration_measurement/radioEmitting_pulser_calibration/.nfs000000000c15ef04000003e7
5940cbffb4ac   41MiB NuRadioReco/examples/test_data.hdf5
d40a7d846c38   47MiB NuRadioReco/examples/cr_efficiency_analysis/output_air_shower_reco/air_shower_reco_pb_80_180_tt_2.7e-05.nur
```

It is of course possible to do a shallow checkout or one with limited history, but this is not what most people will do. This means that on shared computers with many users checking out the repository, a lot of extra disk will be used. Someone doing this in Greenland will take up a ton of bandwidth. 

**Expected behavior**
The git checkout is not much bigger than the size of HEAD. 

**Additional context**

There are a number of ways to remove old large files from the repository (the easiest is probably https://rtyley.github.io/bfg-repo-cleaner/ ), followed by a force push. It's not clear to me though if the large files can easily be accidentally reintroduced by pushes from extant checkouts or not, so maybe this is not easy to fix.  Probably some sort of push hook can somehow prevent that?  Perhaps at least we can document a slimmer checkout?

sjoerd-bouma, 2025-02-27:
Having had a brief look at this... I don't think fixing this is a good idea. While it would be 'nice' to have a smaller repository, the use cases where 500 MB are prohibitive are really exceptional ('fresh clone of NuRadioMC while at Summit Station in Greenland') and are much better covered by educating deploying students (similar to how they are hopefully told not to stream basketball matches while out there). If for some reason they did not think to install NuRadioMC before but suddenly decide that they want to start using (and contributing to!) NuRadioMC while at Summit, they can be told to use a shallow checkout. 

I don't see a way to make the alternative (rewriting and force-pushing the whole repository) fool-proof; as you mention, **any** push from someone with the current repository would undo all of the benefits anyway, and while we can provide push hooks to try to prevent this, these would (as long as we use Github) always be user-side and therefore error prone. So I think the only case where we might want to revisit this and 'fix' our history is if/when we move to a new repository.

fschlueter, 2025-05-02:
I will close this issue. @cozzyd feel free to reopen it.

closed 2025-05-02 by fschlueter