nu-radio/NuRadioMC #882: Implement bookkeeping and time logging into channelBlockOffsetFitter

fschlueter, 2025-03-24:
It would be a nice feature if the module would report how often a significant offset was removed and how much time it takes per event.

fschlueter, 2025-05-02:
@sanyukta13 any update?

sanyukta13, 2025-05-02:
Hi Felix,

Waiting for the iglu-cal measurements to finalize the pulse out of
cal_board + iglu_cal and then can be used in simulations to feed into the
tx antenna.

Best Regards,
Sanyukta Agarwal
Contact: +1-785-424-2512
sites.google.com/view/sanyuktaagarwal
<http://sites.google.com/view/sanyuktaagarwal>

sanyukta13, 2025-05-02:
HI Felix, didn't get the time to look into this yet, will look it over the weekend!

sanyukta13, 2025-10-06:
HI @fschlueter , the function in [channelBlockOffsetFitter](https://github.com/nu-radio/NuRadioMC/blob/add-bookkeeping-channelBlockOffsetFitter/NuRadioReco/modules/RNO_G/channelBlockOffsetFitter.py) which prints the timing info is `end()` which is not being called in either of the rno-g data reading modules so it will not be displayed during data reading. I am not sure if you mean it to be that way? 

Data reading modules I am referring to: dataProviderRNOG and readRNOGDataMattak

fschlueter, 2025-10-07:
Yes, please do that!