nu-radio/NuRadioMC #884: Storing traces in Fourier domain

fschlueter, 2025-03-26:
Right now any `BaseTrace` object will be written to a `.nur` file in the time domain. Is there a good reason for this? Allowing to store them in the Fourier domain allow to skip two fft operations and only makes the (de)serialize slightly longer (of course we make sure that we do not store the data twice). But maybe I am missing something

sjoerd-bouma, 2025-03-26:
What's the use case? It slightly speeds up data writing (and data access) for data where you're only interested in the frequency spectrum?

My 2 cents - I don't think we want to worry too much about speed of data writing, and data reading is only sped up if the future user that reads the data wants it in the frequency domain, otherwise it **slows down** data reading. So for consistency and simplicity I think the current implementation is fine.

CamphynR, 2025-03-26:
It would be convenient for me since I'd like to store a lot of filtered frequency spectra for the thermal noise analysis (on an event per event basis). We want to avoid performing a lot of FFTs in this case.

cg-laser, 2025-03-27:
we discusses this in the inicemc call, and the main reservation is that the frequency domain representation is ambiguous and depends on our FFT definition. Problems will arise if we change the FFT definition in the future (it already happened once) or if we change the data output format, which can be read stand-alone. 
Did you benchmark how much time you actually save by this?

fschlueter, 2025-08-08:
I will close this Issue as the corresponding PR is also closed

closed 2025-08-08 by fschlueter