nu-radio/NuRadioMC #885: channelSinewaveSubtraction can cut too harshly in data

CamphynR, 2025-03-26:
When running the channelSineWaveSubtraction with default settings (peak_prominence=4) for a thermal noise analysis we noticed that the module can sometimes cut too much into data as shown on the plots below. The plots show the spectral amplitude distributions for ~300 MHz without and with the CW filtering applied. The left hand side is no CW with ~150k forced trigger events, the right hand side shows the CW filter applied on ~300k forced trigger events. We noticed that the CW filter can cut the tail of these distributions.

[image]

In another use case we were plotting the average spectra for an upward facing LPDA (channel 13) and noticed that the subtraction can overcut on the weather balloon line (403 MHz) as shown on the plot below. The blue lines are the data the other lines can be ignored. 

[image]

alisanozdrina, 2025-03-26:
@CamphynR thanks for bringing this up. Can you maybe send me some examples: station, run and event numbers where you see over subtraction happening, so I check possible solutions? 

Also, I added new peak finding algorithm (based on sliding window RMS calculation), I will create a new branch to test this and other things.

CamphynR, 2025-04-09:
Hi Alisa thanks for the feedback. I can already show a set of around 10k events that was ran with a peak_prominence=10. This is for season2023 station23 channel 13 in the beginning of the season (so runs 1136, 1137, 1138, 1139, etc). You can see that the overcutting is solved but the weatherballoon line around 400 MHz is visible. It also fails at cutting away the peak at 200 MHz. I tried out the new algorithm but I failed to get enough statistics since it increases the run time by a very serious amount (5 seconds per event with the 'sliding' algorithm vs <1s with the 'simple' algorithm). I should also note that these problems of overcutting are only visible in the LPDA channels. Let me take some time to get the new algorithm to work a bit faster.

[image]

I tested the new algorithm on station 23, run 1139, event 132, channel 13 with peak_prominence=4, you can see that the peak at around 160 MHz is removed with quite a deep cut and other peaks are not quite picked up.

[image]

alisanozdrina, 2025-04-10:
Hi Ruben, 

I’m surprised it takes so long to run a single event on your setup. I checked the execution time on my machine, and it’s 0.135945/0.761079 for the simple and sliding algorithms. Is the 1 to 5 seconds just for the sine subtraction, or does that include the RNO-G reader, calibration, and other stuff?

CamphynR, 2025-04-14:
Hi Alisa. The comparison is with the rnogDataProvider included in both times (blockoffset remover, cable delay and glitch detection). Let me check whether it maybe gets confused on events due to one of these operations. I'm also quite surprised on the difference so maybe I can do some further investigations.

fschlueter, 2025-08-08:
Hi @CamphynR and @alisanozdrina is that still open? @CamphynR what setting did you finally use? Did you check the rayleigh profiles of the forced trigger to see if they are "cut"?