nu-radio/NuRadioMC #899: channelReadoutWindowCutter introduces small artifacts at low frequencies

cg-laser, 2025-04-07:
This is probably of no concern for most users, but we found that cutting traces results in artifacts in the Fourier domain. Previously filtered frequency bins are no longer zero:

[image]

[image]

It is a small effect, and the plot below is zoomed in substantially, but we ran into this problem when doing an analysis in the Fourier domain, where low-frequency bins no longer agree with the model prediction. 

The effect can be reduced by filtering the trace in the time domain first using e.g. a Hann window, i.e., replacing 
`trace = trace[:number_of_samples]`

with 
```
from NuRadioReco.utilities import signal_processing
window = signal_processing.half_hann_window(number_of_samples, half_percent=0.05)
trace = channel.get_trace()
trace = trace[:number_of_samples] * window
```

fschlueter, 2025-05-02:
Are you proposing a PR for this?

sjoerd-bouma, 2025-05-07:
Is this really an issue, or just something to be aware of? My understanding is that the channelReadoutWindowCutter simulates the finite readout window of an actual detector experiment, and the above would be expected in real data, too.

fschlueter, 2025-08-08:
@cg-laser ?