nu-radio/NuRadioMC #907: Cannot deploy to PyPI with (optional) github dependencies

sjoerd-bouma, 2025-04-14:
Currently the deployment to PyPI fails because we have several dependencies that install from Github (instead of PyPI). They are all optional dependencies, currently:

- pygdsm (needed for galactic noise simulations)
- nifty, pypocketfft (needed for IFT energy reconstruction)
- mattak (for reading RNO-G data)
- rnog-runtable (optional, only for RNO-G run table support, only works for RNO-G authorized members)
- cr-pulse-interpolator

Our options are to either switch to PyPI versions of the above modules (which may work for some, but certainly not for all; we don't control pygdsm, nifty and pypocketfft); switch to a different mechanism to install (some) optional dependencies (e.g. add an additional `optional_requirements.txt` file); or remove these optional dependencies for the build that is published to PyPI only. Other, cleverer suggestions are welcome

fschlueter, 2025-04-14:
How would that work? Are there different rules for dependencies which are listed in a `opetional_requirements.txt` file than for those directly listed in the toml file? I tried to read up on this in the [pypi docu](https://packaging.python.org/en/latest/specifications/core-metadata/#core-metadata-provides-extra) but did not get far.

sjoerd-bouma, 2025-04-14:
> How would that work? Are there different rules for dependencies which are listed in a `opetional_requirements.txt` file than for those directly listed in the toml file? I tried to read up on this in [...]

`optional_requirements.txt` or whatever would just be a text file in the usual pip format, so we can do whatever we like with that, but it will be ignored by poetry and pip (in particular, there's no way to use the convenient `[extras]` option work when calling `pip install nuradiomc[extras]` for these dependencies). So we would have to rely on users calling `pip install -r optional_requirements.txt` (which would also install ALL optional dependencies, no way to select some and ignore others.

The alternative is to keep things as they are on github, but write a script that modifies the pyproject.toml before calling `poetry build` or removes the illegal dependencies from the built wheel before pushing to PyPI. But this will create a difference between github and pypi releases :shrug:.

sjoerd-bouma, 2025-05-01:
Closed (for now) via #914 - maybe we should still document the new `optional_requirements.txt` file better, and if there's a new `pygdsm` release I would like to move that back to the `pyproject.toml` as well. Maybe I'll go bug Danny again : )

PR #914 (2025-04-30): Move optional GitHub requirements
As discussed in the call today, this PR moves the dependencies
```
pygdsm (needed for galactic noise simulations)
nifty, pypocketfft (needed for IFT energy reconstruction)
mattak (for reading RNO-G data)
rnog-runtable (optional, only for RNO-G run table support, only works for RNO-G authorized members)
```
out of the pyproject.toml and into a separate `optional_requirements.txt` file. I've optimistically assumed that the `cr-pulse-interpolator` will get an updated PyPI release soon.

This closes #907
closed 2025-05-01 by sjoerd-bouma