nu-radio/NuRadioMC #915: Properly vectorizing attenuation.get_attenuation_length

fschlueter, 2025-04-17:
Currently you can either pass a vector of frequencies or a vector of depths to `attenuation.get_attenuation_length` but not both together (the other has to be an float). 

MWE:
```
from NuRadioMC.utilities import attenuation
import numpy as np

for model in attenuation.model_to_int:
    print(model)
    try:
        print(attenuation.get_attenuation_length(np.array([-100, -200]), np.arange(0.1, 1, 0.1), model))
    except Exception as e:
        print(f"Error: {e}")
```

sjoerd-bouma, 2025-05-04:
I'm confused, what's the expected outcome here? An outer product (2 x 3 array)? I would have expected this to work if and only if the two arrays cast to compatible shapes.

Testing this tells me
1) using two arrays that are the same shape work (e.g. depths (N,M) & frequencies (N,M) -> attenuation_length (N,M)), which is actually what I would expect;
2) using array shapes that are different but could numpy-cast together does not work (e.g. (1,N) & (M,1) raise an error instead of casting to (M,N) as might be expected)

I think the fact that 2) does not work... is probably fine? If this is a feature you think we should support, feel free to reopen this, but the MWE you posted does not look like a bug/problem to me.

fschlueter, 2025-05-05:
Well its not a but but it would be a great feature to have! In the raytracer we are currently using a lot of comprehensive lists to calculate the depth and frequency dependent attenuation length. I expect the code to speed up if we changed that. And I think you first point offers an elegant way to archive that.

fschlueter, 2025-05-05:
I also did not label it as a bug?

sjoerd-bouma, 2025-05-05:
Okay, sure, though in that case the issue title is slightly misleading IMO, and certainly your MWE should always continue to raise an error, so you might want to update it & the description.

sjoerd-bouma, 2025-05-05:
You have an `(2,)` and a `(9,)` shaped array, what do you expect? Numpy would also complain if you try to perform some vector operation with those two. `(2,)` and `(2,)` already works, the only thing that's 'missing' is support for unequal array shapes but where the shapes can be cast to be compatible, e.g. `(2, 1,)` and `(1, 9,)`. But this is not what you have in your MWE.

or in case this wasn't clear in my previous example: 1) **already works**, and I don't think 2) is useful enough to spend time on.

fschlueter, 2025-05-05:
Sjoerd, you do not see the appeal of being able to get the attenuation length for a (2,) and a (9,) shaped array, fine. I do think that would be very useful feature. What is your problem?

fschlueter, 2025-05-05:
> If I take two arrays of the same shape (N,M) and (N,M) you're arguing this should return something that's (N,M,N,M)-shaped?

come on...

fschlueter, 2025-05-05:
So my intention behind this issue is to suggest the following "feature":
If we pass a depth vector with the shape `(N,)` and a frequency vector with shape `(M,)` that the code returns a attenuation_length vector of shape `(N, M)`. I think that would be a very natural interface for the function. Plus it would make the following code easier:
```
                    attenuation_factor_exponent_tmp = np.array(
                        [[dt(x, C_0, f) * dx_actual for x, dx_actual in zip(mid_points, dx_actuals)]
                        for f in freqs])
```

fschlueter, 2025-05-07:
In the use case 4, wouldn't the current output be wrong?