nu-radio/NuRadioMC #919: New io_utilities break pickling numpy arrays containing objects and dictionaries

MartinRavn, 2025-04-24:
I get a segmentation fault with no error message if I try to save a numpy array containing objects or dictionaries after importing NuRadioMC. This happens because of the new `_pickle_numpy_array` and `_unpickle_numpy_array` in `NuRadioReco/utilities/io_utilities.py` that overwrite the standard pickling mechanism for numpy arrays. I have made this minimal script that can be run with `mode = "save"` or `mode = "load"` that reproduces the error:

```
import numpy as np
import pickle
import NuRadioMC

mode = "save"

filename = "test_pickle.pkl"

if mode == "save":
    results = np.zeros([5,10], dtype=object)
    with open(filename, "wb") as f:
        pickle.dump(results, f)

elif mode == "load":
    with open(filename, "rb") as f:
        results = pickle.load(f)

print(results)
```

fschlueter, 2025-04-24:
Can we catch that and make a warning?

sjoerd-bouma, 2025-04-24:
Yeah, for NuRadio at least I didn't see a reason to worry about supporting numpy arrays of objects, but you're probably right that it's a bit annoying that we overwrite the pickling also outside of NuRadio events. This shouldn't be too hard to fix (it will just require some more code in all places where we serialize stuff), but as Felix mentioned it may have to wait a while.

fschlueter, 2025-04-25:
@MartinRavn I am a bit confused, your MWE works for me

MartinRavn, 2025-04-25:
It fails for me if I first run it with `mode = "save"` and then `mode = "load"`. I have tried with both numpy 1.26.2 and 2.0 and on both our server and my local computer. I also just pulled the develop branch.

cg-laser, 2025-05-08:
I think this is an extremely critical bug because it changes pickle everywhere when importing NuRadioMC, i.e., if you include `import NuRadioMC` somewhere in your code, it changes the pickle behaviour everywhere. 
If we make custom hacks to pickle, this should be contained to the nur files only.