nu-radio/NuRadioMC #938: 0 event station causing list index out of range error in Simulation.py

SuedIsProgramming, 2025-05-30:
**Describe the issue**
I am attaching a zipped file with my simulation code: `step2_run_simulation`. In it, I use `station.json`.
[issue_code.zip](https://github.com/user-attachments/files/20527141/issue_code.zip)

However, when I run the code I get the following error: 

Traceback (most recent call last):
  File "/data/condor_shared/users/ssued/tutorials-rnog/plot_sim_event/step2_run_simulation.py", line 67, in <module>
    sim.run()
  File "/data/condor_shared/users/ssued/RNOG_Image_Builder/venv/lib/python3.11/site-packages/NuRadioMC/NuRadioMC/simulation/simulation.py", line 1722, in run
    self._output_writer_hdf5.add_event_group(output_buffer)
  File "/data/condor_shared/users/ssued/RNOG_Image_Builder/venv/lib/python3.11/site-packages/NuRadioMC/NuRadioMC/simulation/output_writer_hdf5.py", line 390, in add_event_group
    evt = stn_buffer[list(stn_buffer.keys())[0]]
                     ~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range

**Expected behavior**
The expected behavior is for the simulation to run and return a working (or empty) .nur file.

**Additional context**
I believe I have narrowed down the issue to be in simulation.py from lines 1495 to 1722 in simulation.py, I believe the issue is caused because a dictionary with entries '101' and '102' are made, but the if statement in line 1566 eliminates all events from station 101. So in the output buffer I simply have:

output_buffer = {101: {}, 102: {0: <NuRadioReco.framework.event.Event object at 0x7f0bbcdc9e90>}}

When this is sent into output_writer_hdf5.py, it errors out because there is no zeroth element inside of station 101 and thus the indexError.

@sjoerd-bouma (Sorry for tagging I was told you would be capable to help me with this)

sjoerd-bouma, 2025-06-03:
Hi Santiago, thanks for flagging this, unfortunately our quickest first responder is in Greenland and I'm at a LOFAR meeting, hence the delay. But thanks to your very detailed debugging I think #942 should fix this (you can try merging this branch into whichever branch your currently working on to fix the issue, or otherwise it should get merged hopefully soon).

SuedIsProgramming, 2025-06-04:
Ok! Thank you for your help. I am currently busy and have done a lot of changes to my code to test it again immediately, but I will follow up in the following days.

closed 2025-06-06 by sjoerd-bouma